peterhinch/micropython-micro-gui #2: Support I2C keypad

petrkr, 2022-01-06:
Thinking how I can use input from external libraries like https://github.com/octopuslab-cz/micropython-keypad

maybe if it can handle some callbacks or I have to learn asyncio already to use Event() or something. Maybe we can somehow use/merge it... 

I have that keypad in alpha state which "now just works", but I can rewrite it. I want to support interrupt than polling, but keep polling, because not always you have connected IRQ pin

peterhinch, 2022-01-07:
I think the keypad driver would need to be modified to use `uasyncio`. Currently its `read` and `input` methods block, which would prevent display updates.

petrkr, 2022-01-07:
Yes I know.. For this case I would use non-blocking `get_key` which can be ran in while True loop and if no key pressed, then it returns `None`.

`read` and `input` was designed to use with "spaghetti" code style for fast proof-of-concepts, also it call back "on_keypress" event, so I did tft refresh (or draw) in such callback.

I have to learn more about asyncio method of coding, but I still want to keep some "simple" blocking way so beginner programmer do not need handle callbacks etc, but for cost of blocking reads

petrkr, 2022-01-11:
For now I wrapped my i2c keypad to emulated class and it works as proof-of-concept right.

```
class KeypadPin():
    def __init__(self, keypad, key, invert = False):
        self._kp = keypad
        self._inv = invert
        self._key = key

    def value(self, value = None):
        if value is not None:
            # we do not support set value here
            return

        char = self._kp.get_key()
        if not char:
            return 0 ^ self._inv
        
        if char == self._key:
            return 1 ^ self._inv
        else:
            return 0 ^ self._inv

    def __call__(self, value = None):
        return self.value(value)
```

petrkr, 2022-01-12:
I think I am happy with this implementation, closing

closed 2022-01-12 by petrkr