peterhinch/micropython-micro-gui #11: Feature: text align

petrkr, 2022-04-07:
I think calculate some text align will not be easy part. 

I have some labels as "status" messages, and those will be handy to have in center, but problem is status messages have different length, so fixed position is not good.

peterhinch, 2022-04-08:
The `Label` widget supports only left alignment. However you can display text using the `Writer` or `CWriter` object - the `stringlen` method returns the length of the string in pixels. You can therefore determine the start location for centre alignment. One approach might be to subclass `Label` although I haven't considered this in detail.

peterhinch, 2022-04-08:
I have pushed an update: the `Label` widget now supports a `justify` constructor arg allowing left, centre and right justification. Thank you for the suggestion: it proved easy to implement and is a worthwhile enhancement.

closed 2022-04-08 by peterhinch