peterhinch/micropython-micro-gui #12: Feature/Question: Set default active input on screen

petrkr, 2022-04-07:
Will be handy to select which input will be default after screen shows up.. Now it's most left button or widget.

But I have buttons layout (in HW) as SWITCH and SELECT. and OnScreen I have "No, Yes" buttons, but No is selected as first. I know I can swap on-screen buttons, but would be handy to pre-select that "YES" one, which is right (same as SELECT/ENTER button is on right side)

peterhinch, 2022-04-08:
See [this part of the doc](https://github.com/peterhinch/micropython-micro-gui/blob/main/README.md#tab-order-and-button-layout). You have full control over which object is the default and the order in which they are subsequently selected.

petrkr, 2022-04-08:
Oh yea, that works. I swap also creating of elements to make it "logically" in code from left-->right order.. So yes, if I will initialize first item which I want to be default, then it works.

I read doc, but not too carefully to find this section. I look for it around widget / buttons sections

closed 2022-04-08 by petrkr