peterhinch/micropython-micro-gui #13: Cannot install or add directories- need support with ssd1306

johnr1245, 2022-06-20:
Hi there. I am an extreme beginner with python and programming generally and am trying to use a ssd1306 display with a tiny pico board and cannot for the life of me figure out how to install this as there are few visuals and I am getting extremely confused by the instructions.

 I can open the hardware setup file and attempt to make it (depsite the fact I dont even know if it works as there are very few examples using the sdd1306 and doesnt really explain what the different lines of code mean) As well how is one supposed to get the gui and driver folders into the boards directory as from the IDE im using you cannot do that unless you open every .py file in each of those folders and indivually save each to the directory from within the IDE?  

As well the $ mpremote mount . command does nothing for me when I try to use it in the command line after editing the hardware set up file and nothing in the REPL so I really just have no idea what is supposed to be using what, kinda just says to do it. Really great info in the rest of this but I could really use some clarification in how it is actually installed. Any help would be much appreicated and I can provide photos and what not of the pin lay out and code I have already.  Thank you for all the work!

johnr1245, 2022-06-21:
So I changed some stuff around and got all the folders on the board. I have cleared most of the errors (I think) but am still left with this one which I cannot find any folders or files matching the name. Any idea what this error code means?

[image]

peterhinch, 2022-06-21:
There is something wrong with your `hardware_setup.py` file. It's impossible for me to figure out what the problem is without seeing the code.

johnr1245, 2022-06-21:
My apologies very true lol still learning the formatting when asking questions, thank you very much for the reply my apologies if my initial question was scattered. 

Below is the code below for the `hardware_setup.py` I currently have the navigation button's code omitted as I was sure how to add them to the example code for the ssd1306 but am not sure if that could be causing an issue as well? I have asked on other forums and they had suggested that I am importing `ssd` incorrectly and should be importing it from `gui.core.ugui` instead of `hardware_setup.py`, but I thought the hardware_setup is what defined the `ssd` value for all other files. 

As well this code below is just the example supplied from the nano_gui example for the ssd1306 driver adjusted to the pins for my Tiny Pico, beyond that I have not edited it. Also just out of curiosity, why is there a letter Y before the IO pin number, I tried without and it seemed to have messed with the code so I assumed it to be necessary.

`import machine
from drivers.ssd1306.ssd1306 import SSD1306_SPI, SSD1306_I2C

WIDTH = const(128)
HEIGHT = const(64)
use_spi = False  # I2C or SPI
soft = True  # Soft or hard I2C/SPI

# Export an initialised ssd display object.
if use_spi:
    # Pyb   SSD
    # 3v3   Vin
    # Gnd   Gnd
    # X1    DC
    # X2    CS
    # X3    Rst
    # X6    CLK
    # X8    DATA
    pdc = machine.Pin('Y1', machine.Pin.OUT_PP)
    pcs = machine.Pin('Y2', machine.Pin.OUT_PP)
    prst = machine.Pin('Y3', machine.Pin.OUT_PP)
    if soft:
        spi = machine.SPI(sck=machine.Pin('Y6'), mosi=machine.Pin('Y8'), miso=machine.Pin('Y7'))
    else:
        spi = machine.SPI(2)
    ssd = SSD1306_SPI(WIDTH, HEIGHT, spi, pdc, prst, pcs)
else:  # I2C
    # Pyb   SSD
    # 3v3   Vin
    # Gnd   Gnd
    # Y9    CLK
    # Y10   DATA
    if soft:
        pscl = machine.Pin('Y22', machine.Pin.OPEN_DRAIN)
        psda = machine.Pin('Y21', machine.Pin.OPEN_DRAIN)
        i2c = machine.SoftI2C(scl=pscl, sda=psda)
    else:
        i2c = machine.I2C(2)
    ssd = SSD1306_I2C(WIDTH, HEIGHT, i2c)`

peterhinch, 2022-06-22:
I think the IDE tools you are using may not be showing all the error messages. You might try using the official [mpremote](https://docs.micropython.org/en/latest/reference/mpremote.html). In any support query it's best to cut and paste the full traceback and I don't think you're getting that at the moment. Your setup file will have thrown an exception when it hit the first pin number with an "X" or "Y" designation.

Hardware I2C buses have a number and are usually tied to specific pins. These vary between devices. Alternatively you can use [SoftI2C](https://docs.micropython.org/en/latest/library/machine.I2C.html?highlight=softi2c#machine.SoftI2C) which lets you choose the associated pins.

johnr1245, 2022-06-22:
Just tried mpremote mount and got something different this time that might be the tracebacks you needed to be. Looks pretty messed up. Should I just be wiping my microcontroller and starting again or using a different IDE? Here is a paste bin for traceback: https://pastebin.com/PRjABZFW

peterhinch, 2022-06-23:
This seems to be a problem with Thonny. I've never used this, so can't help. If you move to [the forum](http://[forum](http://forum.micropython.org/index.php).micropython.org/index.php) the developer of Thonny is available to help.

johnr1245, 2022-06-23:
Alright will try. Should I just try all the steps again on a different IDE like Pycharm or VScode? Seems like alot of the beginner IDEs have pretty poor error reporting.

peterhinch, 2022-06-24:
In MicroPython I am a command-line guy - I use mpremote or rshell. For advice on IDE's please ask in the forum.

peterhinch, 2023-02-23:
Closing due to inactivity.

closed 2023-02-23 by peterhinch