peterhinch/micropython-micro-gui #15: SSD1306 driver is reported as incompatible device driver.

DusKing1, 2022-07-07:
Hi, thanks for your great job! I really like this micro-gui framework.

But when I'm trying to use it on a 128x64 SSD1306 I2C display, I got: 
```
  File "main.py", line 34, in <module>
  File "main.py", line 32, in test
  File "gui/core/ugui.py", line 305, in change
  File "main.py", line 19, in __init__
  File "gui/core/writer.py", line 261, in __init__
OSError: Incompatible device driver.
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
Type "help()" for more information.
```

I copied all necessary `.py` scripts (I think), my file structure is like this:
[image]

My `hardware_setup.py` is 
```
from machine import Pin, I2C, freq
import gc
from drivers.ssd1306.ssd1306 import SSD1306_I2C as SSD
freq(240_000_000)
i2c = I2C(0, scl=Pin(22), sda=Pin(21), freq=1_000_000)
gc.collect()
ssd = SSD(128, 64, i2c)
gc.collect()

from gui.core.ugui import Display, quiet

nxt = Pin(34, Pin.IN)  # Move to next control
sel = Pin(32, Pin.IN, Pin.PULL_UP)  # Operate current control
prev = Pin(35, Pin.IN)  # Move to previous control
increase = Pin(33, Pin.IN, Pin.PULL_UP)  # Increase control's value
decrease = Pin(25, Pin.IN, Pin.PULL_UP)  # Decrease control's value
display = Display(ssd, nxt, sel, prev, increase, decrease)
```
and my `main.py` is:
```
import hardware_setup  # Create a display instance
from gui.core.ugui import Screen, ssd

from gui.widgets import Label, Button, CloseButton
from gui.core.writer import CWriter
# Font for CWriter
import gui.fonts.arial10 as arial10
from gui.core.colors import *

class BaseScreen(Screen):

    def __init__(self):

        def my_callback(button, arg):
            print('Button pressed', arg)

        super().__init__()
        wri = CWriter(ssd, arial10, GREEN, BLACK, verbose=False)

        col = 2
        row = 2
        Label(wri, row, col, 'Simple Demo')
        row = 50
        Button(wri, row, col, text='Yes', callback=my_callback, args=('Yes',))
        col += 60
        Button(wri, row, col, text='No', callback=my_callback, args=('No',))
        CloseButton(wri)  # Quit the application

def test():
    print('Simple demo: button presses print to REPL.')
    Screen.change(BaseScreen)  # A class is passed here, not an instance.

test()
```

DusKing1, 2022-07-07:
NVM, I have a quick fix in my PR.

peterhinch, 2022-07-08:
Thanks for the clear report. The **CWriter** class is intended for color displays with **Writer** being the monochrome equivalent.

While your mod will probably work fine, the way it is supposed to be used with a monochrome display is as follows. In **main.py**:
```python
# from gui.core.writer import CWriter
from gui.core.writer import Writer  # Monochrome writer
```
and in your code wherever you need a **Writer**:
```python
wri = Writer(ssd, arial10, verbose=False)  # No colors specified
```
Please could you try this and report back. I'll review the docs.

Incidentally you don't need to copy the entire drivers tree to your device - although the unused drivers only consume Flash space.

peterhinch, 2022-07-08:
I've updated the example code in `README.md` to clarify usage with monochrome displays.

DusKing1, 2022-07-08:
Thank you for your reply, I'll test it according to your suggestion once I back to my test hardwares, like next Thursday or later.

DusKing1, 2022-07-17:
Hi, I just tested on my own hardware and your solution works for me. Related PR closed. Thank you for your awesome work!

closed 2022-07-17 by DusKing1