peterhinch/micropython-micro-gui #24: Having error File "gui/core/ugui.py", line 420, in __init__ AttributeError: 'NoneType' object has no attribute 'height'

AkshatChauhan18, 2023-01-26:
# checkbox.py Minimal micro-gui demo showing a Checkbox updating an LED.

# Released under the MIT License (MIT). See LICENSE.
# Copyright (c) 2021 Peter Hinch

# hardware_setup must be imported before other modules because of RAM use.
import hardware_setup  # Create a display instance
from gui.core.ugui import Screen

from gui.widgets import CloseButton, Checkbox, LED
from gui.core.writer import Writer

# Font for CWriter
import gui.fonts.arial10 as arial10
from gui.core.colors import *

class BaseScreen(Screen):

    def __init__(self):

        super().__init__()
        wri = Writer(ssd, arial10, verbose=False)
        col = 2
        row = 2
        self.cb = Checkbox(wri, row, col, callback=self.cbcb)
        col+= 40
        self.led = LED(wri, row, col, color=YELLOW, bdcolor=GREEN)
        CloseButton(wri)

    def cbcb(self, cb):
        self.led.value(cb.value())

def test():
    print('Checkbox demo.')
    Screen.change(BaseScreen)

test()

while running this code I am getting an error:

Traceback (most recent call last):
  File "<stdin>", line 38, in <module>
  File "<stdin>", line 36, in test
  File "gui/core/ugui.py", line 328, in change
  File "<stdin>", line 22, in __init__
  File "gui/core/ugui.py", line 420, in __init__
AttributeError: 'NoneType' object has no attribute 'height'

peterhinch, 2023-01-26:
Please show me your `hardware_setup.py`.

AkshatChauhan18, 2023-01-27:
I am using ssd1306 oled display therefore i used this code for hardware_setup.py
```
from machine import Pin, SoftI2C
import gc

from drivers.ssd1306.ssd1306 import SSD1306_I2C as SSD

i2c = SoftI2C(scl=Pin(22), sda=Pin(21))

oled_width = 128
oled_height = 64
gc.collect()  # Precaution before instantiating framebuf
ssd = SSD(oled_width, oled_height, i2c)

```

peterhinch, 2023-01-28:
Your setup file appears to be designed for `nanogui`. 

In `microgui` you need to define your control buttons and create a `display` object. Please study the docs and the `microgui` setup examples.

peterhinch, 2023-02-23:
Closing due to inactivity.

closed 2023-02-23 by peterhinch