peterhinch/micropython-micro-gui #31: Sharp Display SPI RP2040

wfwAnthony, 2023-05-01:
It seems that there is an issue initializing the SPI in the sharp driver on the RP2040. I've tried switching to SoftSPI for the SPI.LSB with no success.

I'm not too experienced with MicroPython so I may looking in the wrong spot. 

Possibly adding a setup example for the Sharp displays could help.

peterhinch, 2023-05-02:
SPI on RP2040 works: I have used it extensively with various displays. I recommend specifying the pins to use as the defaults are undocumented. Please could you post your `hardware_setup.py`.

wfwAnthony, 2023-05-02:
```
from machine import Pin, SPI, freq
import gc

from drivers.sharp.sharp import SHARP as SSD
freq(250_000_000)  # RP2 overclock
# Create and export an SSD instance
pcs = Pin(2, Pin.OUT, value=1)
spi = SPI(0, sck=Pin(6), mosi=Pin(7), miso=Pin(4))
gc.collect()  # Precaution before instantiating framebuf
ssd = SSD(spi, pcs)

from gui.core.ugui import Display
# Create and export a Display instance
# Define control buttons
nxt = Pin(19, Pin.IN, Pin.PULL_UP)  # Move to next control
sel = Pin(16, Pin.IN, Pin.PULL_UP)  # Operate current control
prev = Pin(18, Pin.IN, Pin.PULL_UP)  # Move to previous control
increase = Pin(20, Pin.IN, Pin.PULL_UP)  # Increase control's value
decrease = Pin(17, Pin.IN, Pin.PULL_UP)  # Decrease control's value
display = Display(ssd, nxt, sel, prev, increase, decrease, 4)  # Encoder

```
I receive an error

`NotImplementedError: LSB`

```
from machine import Pin, SoftSPI, freq
import gc

from drivers.sharp.sharp import SHARP as SSD
freq(250_000_000)  # RP2 overclock
# Create and export an SSD instance
pcs = Pin(2, Pin.OUT, value=1)
spi = SoftSPI(sck=Pin(6), mosi=Pin(7), miso=Pin(4), baudrate=2_000_000, firstbit=machine.SoftSPI.LSB)
gc.collect()  # Precaution before instantiating framebuf
ssd = SSD(spi, pcs)

from gui.core.ugui import Display
# Create and export a Display instance
# Define control buttons
nxt = Pin(19, Pin.IN, Pin.PULL_UP)  # Move to next control
sel = Pin(16, Pin.IN, Pin.PULL_UP)  # Operate current control
prev = Pin(18, Pin.IN, Pin.PULL_UP)  # Move to previous control
increase = Pin(20, Pin.IN, Pin.PULL_UP)  # Increase control's value
decrease = Pin(17, Pin.IN, Pin.PULL_UP)  # Decrease control's value
display = Display(ssd, nxt, sel, prev, increase, decrease, 4)  # Encoder

```

When replacing with SoftSpi  to try and get around the LSB I get 

`ValueError:firstbit must be MSB`

peterhinch, 2023-05-03:
I can replicate this. You have indeed found a bug in the RP2 implementation of SPI. The Sharp displays require LSB-first implementation of SPI and, as you have discovered, this is unsupported. (My testing was with STM, before RP2 was available).

I will raise an issue.

peterhinch, 2023-05-03:
I'm afraid we're in the hands of the maintainers now.