peterhinch/micropython-micro-gui #32: Question: Is there a way to dynamically update the elements listed in a Dropdown object?

evan-rasmussen, 2023-05-19:
Title pretty much explains it. I am trying to get it so the user can input a string that then needs to be added to the elements in a dropdown object. Is there a way to do this dynamically?

peterhinch, 2023-05-20:
That is an interesting question, but I'm very short of time today so this answer is brief and might be described as an educated guess. You might like to experiment.

The dropdown instance has a bound variable `.elements` which holds the text values. When the user clicks on the dropdown, a listbox is instantiated [here](https://github.com/peterhinch/micropython-micro-gui/blob/3de3a8cec8c8b102c590bfea9f9e2e80c5d9812c/gui/widgets/dropdown.py#L28) using those elements (see docs for listbox). So it might be possible to modify `dropdown.elements` at times when the dropdown is not open. See [listbox docs](https://github.com/peterhinch/micropython-micro-gui/blob/main/README.md#67-listbox-widget) for the two possible formats of `.elements`.

Troyhy, 2024-09-03:
I had similar need and I have test implementation for listbox https://github.com/Troyhy/micropython-micro-gui/blob/ESP-Now/gui/widgets/listbox.py#L75-L97 but this is very rough on edges, I did not fully got the scrolling of elements and updating elements right.

peterhinch, 2024-09-04:
This is evidently worth doing. I'll investigate this, but it may be a few days before I get the opportunity