peterhinch/micropython-micro-gui #36: encoder with switch as input?

eudoxos, 2023-09-02:
Hello, currently there is the option to use 3 buttons for navigation. Could I get some idea on how to enhance the code to use +/-/click functionalities of the encoder as next/prev/sel in the Input class? Thanks!

peterhinch, 2023-09-03:
@eudoxos If I understand correctly you want to use an encoder as the only control, rotating it for next/prev and pressing it for sel. This is an interesting idea for projects that don't need increase/decrese. Currently this is not supported: the encoder is used only for adjusting numeric controls although pressing it can be assigned to sel.

It would need some rejigging of the interface and I'm afraid I don't have time to study this further.

eudoxos, 2023-09-04:
OK, I understand this might be disruptive, so I will just put the idea here in more detail and then close the issue:

* use encoder's increment/decrement as previous/next
* use encoder's button click as selection
* options for increasing/decreasing a control:
   * enter/exit with click: navigate to the control (rotating the encoder), enter (click), then rotate to increase/decrease, then click to go back to widget selection (this is what e.g. car interfaces often use and seems the most intuitive)
      * optionally rotate while pressed for precise adjustment
   * turn *while pressed* for increase/decrease, turn for prev/next control
   * opposite of the previous one: turn while pressed for widget prev/next, turn without pressing for increase/decrease

peterhinch, 2023-09-04:
Thanks for that.

It's a mode of operation that I hadn't considered, but it does look promising. On screens with a lot of controls pressing next/prev does get rather tedious. I'll leave the issue open for when I can find time to study it properly.

peterhinch, 2023-09-12:
@eudoxos I have pushed an update which offers this. Documentation to follow.

If you'd like to test it please update `gui/core/ugui.py`. The "encoder only" mode is invoked by changing your `hardware_setup.py` as follows:
```python
sel = Pin(16, Pin.IN, Pin.PULL_UP)  # Encoder button
x = Pin(20, Pin.IN, Pin.PULL_UP)  # Encoder X
y = Pin(17, Pin.IN, Pin.PULL_UP)  # Encoder Y
display = Display(ssd, x, sel, y, False, None, 4)  # Encoder-only mode
```
`incr=False` sets encoder only mode
`encoder=4` defines the encoder division ratio.

When you click on a variable control it goes into adjustment mode. A long press puts it into precision mode.

peterhinch, 2023-09-13:
Thank you very much for this suggestion which works well, and is a great improvement.

I find the interface fast and intuitive. It is especially effective when there are a lot of widgets on a screen. Closing this as complete, but comments are of course welcome.

closed 2023-09-13 by peterhinch