peterhinch/micropython-micro-gui #37: Unable to suppress drawing of the border around the buttons

felis, 2023-10-18:
Specifying `bdcolor=False` or `bdcolor=BLACK` in `Button` doesn't seem to work. The border around the button still appears. Similarly, the border around a `Label` won't appear when I specify `bdcolor=True`.  Modified `simple.py` is here ->  https://gist.github.com/felis/6ed1a85f4d7129b9ec67971e3215153b . The board is Lilygo TTGO T-Display V.1.1, the Micropython binary is ESP32_GENERIC-20231005-v1.21.0.bin .

I can post pictures if necessary.

Thank you,
Oleg.

peterhinch, 2023-10-19:
Hi Oleg,
taking your second problem first - the `Label` - specifying `bdcolor=True` is invalid. Acceptable values are `False`, `None` or a color (see docs). As far as I can see `Label` is behaving correctly.

There is a problem with `Button`. What seems to be happening is this. The `border` arg is behaving correctly in that it controls a border which is 2 pixels outside the dimensions of the `Button`. However the `Button` itself is being rendered incorrectly with its own border at its exact dimensions. This extra border can't be controlled. I will investigate and report back.

Regards,
Pete

peterhinch, 2023-10-19:
Oleg - I hope this answers your queries. If not please reopen or open a new issue.
Regards,
Pete

closed 2023-10-19 by peterhinch

felis, 2023-10-19:
Pete,

Just checked - it works. Thank you very much!

Oleg.