peterhinch/micropython-micro-gui #44: Waveshare Pico LCD 1.3 Support

AaronMittelmeier, 2024-02-12:
Can you please add support for the Waveshare display 1.3 LCD to the Micro GUI?  I can get it loaded, and even running, but it just stays at a black screen.

peterhinch, 2024-02-12:
Unfortunately I don't have one of these displays.

The problem here is that different manufacturers wire up the ST7789 in different ways,necessitating a dedicated `display` 3-tuple. If you look at [this code](https://github.com/peterhinch/micropython-nano-gui/blob/521c7b553d2321efc741282fddd0e72d597a818a/drivers/st7789/st7789_4bit.py#L32) you will see examples, some contributed by users. You may find that one of these works for your display: try them, e.g. by passing `display=PICO_LCD_2` to the constructor. I suggest starting with this one as it's a Waveshare display.

If none of the provided tuples works, try experimenting, starting with whichever provides the best results. See [drivers doc](https://github.com/peterhinch/micropython-nano-gui/blob/master/DRIVERS.md#33-drivers-for-st7789).

To evaluate a tuple, use the [quick hardware check](https://github.com/peterhinch/micropython-micro-gui/blob/main/README.md#16-quick-hardware-check). If this produces pixel-perfect results you can be sure that the GUI will work.

Please post your results and I will update the driver and doc.

AaronMittelmeier, 2024-02-12:
Sweet! Thanks! I'll give it a gander after work this evening. The 1.14 inch, 240x240 hardware_setup is good enough to get the UI to work on the 1.3 device and 3 of the inputs are functional, so it's close, but not quite there. I'll let you know; thanks again.

On a personal note, I was perusing your GitHub and I just wanted to say that I am grateful for people like you who have contributed significantly to the foundation of work related to micropython and controllers in general. If my buddy and I ever produce more than two devices related to our project, I will personally send you a pair.

AaronMittelmeier, 2024-02-12:
[image]
Here's how it looks right now, just an FYI. Sorry about the picture quality. You know how it is. Anyway, I'll have another peek tonight with fresh eyes. Thanks!

AaronMittelmeier, 2024-02-13:
Greetings Again Mr. Hinch,

You were absolutely right.  The magic combo for the display 3-tuple was (0, 0, 16) for this hardware: https://www.waveshare.com/wiki/Pico-LCD-1.3 .

Hardware test
[image]

GUI Demo
[image]

Thank you again for your hard work and your time.  It will be invaluable to my friend and my pet project.

peterhinch, 2024-02-13:
Excellent, thank you for letting me know. I've added a constant `WAVESHARE_13` to the code and driver doc.

closed 2024-02-13 by peterhinch