peterhinch/micropython-micro-gui #52: Keyboard input?

adrianblakey, 2024-12-11:
Any thoughts about a simple widget to provide character key input? I know it’s very tedious to display a keyboard in a grid and navigate and select specific characters to build a string. Perhaps something simpler? a couple of drop downs a-z, 0-9 to pick chars ...

peterhinch, 2024-12-12:
I have considered producing a widget, but there are so many valid layout variants. One option is qwerty, but do you add a shift key? Do you have number keys, or access numbers by another shift key? Then you have the rectangular layout used on TV screens. This doesn't usually have numbers or lowercase, but does minimise screen area and the number of clicks to navigate. Some users might want only numeric and arithmetic symbols. And so on. However the `grid` widget offers possibilities.

You might like to look at the [calendar demo](https://github.com/peterhinch/micropython-micro-gui/blob/main/gui/demos/calendar.py). This displays the days of a month in a `grid` widget. The user can move the current day with the D+ and D- buttons. This could readily be adapted to a TV-style rectangular keyboard layout. An alternative to changing the color of the current day would be to change the background color. Note that the entire `grid` is re-populated on a month change: this could be used to implement "sticky" shift keys.

adrianblakey, 2024-12-13:
Thank you! I know a comprehensive solution would take a lot of effort, and a lot of code which even on a Pico 2 would be piggish. I’ll look closer at the calendar demo.

peterhinch, 2024-12-13:
I've been experimenting! The following script works on a Pico 2 with a large (480x320) screen.
```py
# keyboard.py Test Grid class.

# Released under the MIT License (MIT). See LICENSE.
# Copyright (c) 2024 Peter Hinch

import hardware_setup  # Create a display instance
from gui.core.ugui import Screen, ssd

from gui.widgets import Grid, CloseButton, Label, Button
from gui.core.writer import CWriter

# Font for CWriter
import gui.fonts.font10 as font
import gui.fonts.font14 as font1
from gui.core.colors import *

class BaseScreen(Screen):
    def __init__(self):

        super().__init__()
        wri = CWriter(ssd, font, GREEN, BLACK, False)
        wri1 = CWriter(ssd, font1, WHITE, BLACK, False)
        col = 2
        row = 2
        self.rows = 4
        self.cols = 10
        self.ncells = self.cols * self.rows
        colwidth = 35
        self.grid = Grid(wri, row, col, colwidth, self.rows, self.cols, justify=Label.CENTRE)
        self.grid[0, 0:10] = iter("1234567890")
        self.grid[1, 0:10] = iter("qwertyuiop")
        self.grid[2, 0:10] = iter("asdfghjkl;")
        self.grid[3, 0:9] = iter("zxcvbnm,.")
        self.grid[3, 9] = "sp"  # Special char: space
        self.cr = 0  # Current grid row and col
        self.cc = 0
        self.grid[0, 0] = {"bgcolor": RED}  # Highlight currency

        row = self.grid.mrow + 4  # Navigation Button instances
        db = {"height": 30, "shape": CIRCLE, "callback": self.adjust}
        b = Button(wri, row, 100, text="up", args=(-1, 0), **db)
        b = Button(wri, row + 40, 60, text="l", args=(0, -1), **db)
        b = Button(wri, row + 40, 100, text="s", args=(0, 0), **db)  # Select
        b = Button(wri, row + 40, 140, text="r", args=(0, 1), **db)
        b = Button(wri, row + 80, 100, text="dn", args=(1, 0), **db)
        row = b.mrow + 5
        col = 2
        self.lbltxt = Label(wri1, row, col, text=ssd.width - 4, fgcolor=WHITE)
        self.lbltxt.value("Text: ")  # Show result
        CloseButton(wri)  # Quit the application

    def adjust(self, _, dr, dc):
        g = self.grid
        cl = next(g[self.cr, self.cc])  # Current Label
        g[self.cr, self.cc] = {"bgcolor": BLACK}
        self.cr += dr
        self.cr %= self.rows
        self.cc += dc
        self.cc %= self.cols
        if dr == 0 and dc == 0:  # Select was pressed
            ct = cl()  # Check special character(s)
            ct = " " if ct == "sp" else ct
            self.lbltxt.value(self.lbltxt.value() + ct)
        g[self.cr, self.cc] = {"bgcolor": RED}

def test():
    print("Keyboard demo.")
    Screen.change(BaseScreen)  # A class is passed here, not an instance.

test()
```
It could readily be adapted to dynamically re-map the grid for upper case, extra symbols etc. However the UI is unpleasant because there is an extra level of indirection: you have to operate `Button` objects to navigate in the grid.

Another option would be a `Grid`-like array of `Button` objects. However micro-gui only supports one-dimensional navigation, so to get from "q" to "m" would require 27 presses of the Next physical button. This might be manageable with an encoder-only interface, but would be tedious with physical buttons.

The final option would be to implement it in the touch GUI only. I think this could be done by overlaying a `Pad` widget on top of a `Grid`. (The `Pad` would require a small modification.) This has much more promise because the touch GUI is not restricted to a sequential "tab order" of controls: the UI would be fast and intuitive.

I'd be interested to hear your comments, on the usability of code designed like the above, and any other observations/ideas.

adrianblakey, 2024-12-13:
V useful ty - despite having to only a st7789 display. I made the fonts smaller to get the grid to fit. I’ll play with it more tomorrow to make the buttons fit. I have the 5 physical navigation buttons so the widget buttons could be dispensed with.

peterhinch, 2024-12-14:
I think you're misunderstanding how this works. The `Grid` is an array of `Label` instances: these are passive widgets and cannot take the focus. The above script works round this by means of on-screen `Button` instances which simulate navigating between cells by changing the background color of the current cell. I'm afraid you're stuck with this awkward means of navigation.

This could be overcome by replacing the `Grid` with an array of `Button` instances: these can take the focus. However the nature of the GUI is that active widgets exist in a linear "tab order" - navigation is inherently one-dimensional, leading to the 27-press problem mentioned above. Having five buttons does not overcome this. In essence micro-gui is like navigating a PC application with Tab and shift-Tab only: best with small numbers of active widgets.

[micropython-touch](https://github.com/peterhinch/micropython-touch/tree/master) has no limitation in the order in which widgets are accessed. My touch-based keyboard works well.

adrianblakey, 2024-12-14:
I figured it out. I made a couple of changes to make it fit on my tiny screen and add 4 states and make it a bit like the iPhone US keyboard.

```
# keyboard.py Test Grid class.

# Released under the MIT License (MIT). See LICENSE.
# Copyright (c) 2024 Peter Hinch

import hardware_setup  # Create a display instance
from gui.core.ugui import Screen, ssd

from gui.widgets import Grid, CloseButton, Label, Button
from gui.core.writer import CWriter

# Font for CWriter
import gui.fonts.arial10 as font
import gui.fonts.arial10 as font1
from gui.core.colors import *

class BaseScreen(Screen):
    def __init__(self):

        super().__init__()
        wri = CWriter(ssd, font, GREEN, BLACK, False)
        wri1 = CWriter(ssd, font1, WHITE, BLACK, False)
        col = 2
        row = 2
        self.sht = False  # Shift state
        self.num = False  # Numeric lock state
        self.rows = 4
        self.cols = 10
        self.ncells = self.cols * self.rows
        colwidth = 15
        self.grid = Grid(wri, row, col, colwidth, self.rows, self.cols, justify=Label.CENTRE)
        self.__set()
        self.cr = 0  # Current grid row and col
        self.cc = 0
        self.grid[0, 0] = {"bgcolor": RED}  # Highlight currency
        row = self.grid.mrow + 2  # Navigation Button instances
        db = {"height": 18, "shape": CIRCLE, "callback": self.adjust}
        b = Button(wri, row, 100, text="up", args=(-1, 0), **db)
        b = Button(wri, row + 20, 60, text="l", args=(0, -1), **db)
        b = Button(wri, row + 20, 100, text="s", args=(0, 0), **db)  # Select
        b = Button(wri, row + 20, 140, text="r", args=(0, 1), **db)
        b = Button(wri, row + 40, 100, text="dn", args=(1, 0), **db)
        row = b.mrow + 5
        col = 2
        self.lbltxt = Label(wri1, row, col, text=ssd.width - 4, fgcolor=WHITE)
        self.lbltxt.value("Text: ")  # Show result
        CloseButton(wri)  # Quit the application

    def __set(self):
        if not self.sht and not self.num:
            self.grid[0, 0:10] = iter("qwertyuiop")
            self.grid[1, 0:10] = iter("asdfghjkl")
            self.grid[2, 1:8] = iter("zxcvbnm")
        elif self.sht and not self.num:
            self.grid[0, 0:10] = iter("QWERTYUIOP")
            self.grid[1, 0:10] = iter("ASDFGHJKL")
            self.grid[2, 1:8] = iter("ZXCVBNM")
        elif self.num and not self.sht:
            self.grid[0, 0:10] = iter("1234567890")
            self.grid[1, 0:10] = iter('-/:;()$&@"')
            self.grid[2, 1:6] = iter(".,?!'")            
        elif self.num and self.sht:
            self.grid[0, 0:10] = iter("[]{}#%^*+=")
            self.grid[1, 0:10] = iter("_\|~<>€£¥.")
            self.grid[2, 1:6] = iter(".,?!'")
        self.grid[2, 0] = "sh"  # Num
        self.grid[2, 9] = "bs"  # Special char: back space
        self.grid[3, 0] = "nu"  # Num
        self.grid[3, 1] = "sp"  # Special char: space
        self.grid[3, 2] = "rt"  # return             

        
    def adjust(self, _, dr, dc):
        g = self.grid
        cl = next(g[self.cr, self.cc])  # Current Label
        g[self.cr, self.cc] = {"bgcolor": BLACK}
        self.cr += dr
        self.cr %= self.rows
        self.cc += dc
        self.cc %= self.cols
        if dr == 0 and dc == 0:  # Select was pressed
            ct = cl()  # Check special character(s)
            if ct == "sp":
                ct = " " 
            elif ct == "sh":   # Shift
                self.sht = not self.sht
                self.__set()
            elif ct == "nu": # Numeric lock
                self.num = not self.num
                self.__set()
            elif ct == "bs":
                v = self.lbltxt.value()
                v = v[:-1]
                self.lbltxt.value(v)
            elif ct == "rt":
                pass # return the value to the parent dislay?                
            else:
                if ct is not None:
                    self.lbltxt.value(self.lbltxt.value() + ct)
        g[self.cr, self.cc] = {"bgcolor": RED}

def test():
    print("Keyboard demo.")
    Screen.change(BaseScreen)  # A class is passed here, not an instance.

test()
```

peterhinch, 2024-12-15:
Congratulations on sorting it out, your code looks good. I'd be interested in your views of the UI. I found it annoying and unintuitive, but maybe you learn it given time. I can't think of a fully satisfactory solution in micro-gui.

I've published the [touch version](https://github.com/peterhinch/micropython-touch/blob/master/gui/demos/keyboard.py). At the moment the re-mapping for case, numeric and symbols is done via a `Dropdown`; space and backspace are done with `Button` instances rather than specific keys. It's unconventional but seems intuitive in use. I also have a one-line text window with prompt and cursor.

Thanks for prompting me to examine this.

adrianblakey, 2024-12-15:
Given the constraints there’s no prefect solution. I am sure this has been a subject UI experts have debated for years, and humans are very good at adapting to imperfect solutions. I am just glad to have a way to input text which might be used very occasionally to say input a WiFi password. 
It might be nice to be able to interact directly with the grid widget without having use separate buttons. So for example we could navigate l, r, u, d and select, in the grid rather than using the buttons that operate on the grid (this was my initial misconception). I any event, thanks for this!

closed 2024-12-15 by peterhinch