peterhinch/micropython-micro-gui #57: Screen with Labels only

chiefenne, 2025-03-23:
Hi there,

I want to make a "splash screen" for my application. So the first screen to show up (for 3-5 seconds) is some text. The text consists of the application name, version, etc. Then I'll switch to the next screen for the different app functionalities.

I thought I'd use "labels" for that, but the i get the error that no active widget is available. I could workaround to use textboxes, but they have a frame which i dont want.

Do i miss something obvious? Any help appreciated.

Andreas

chiefenne, 2025-03-23:
Just by chance I found a solution. There is a DummyWidget exactly for that purpose. To make it work i needed to add fgcolor and bgcolor to the class attributes.

```
class SplashScreen(Screen):
    def __init__(self):
        super().__init__()

        self.fgcolor = WHITE
        self.bgcolor = BLACK

        wri = Writer(ssd, font, verbose=False)
        text = f'SERVO Tester {__version__}'
        Label(wri, 2, 22, text)
        DummyWidget(wri, self)
```

closed 2025-03-23 by chiefenne

peterhinch, 2025-03-23:
This was intended for popup message windows (e.g. "Busy .. please wait") however they are entirely suited to your purpose. I'll think about making the use of a `DummyWidget` more user friendly (along the lines of its use in `Window` objects).

peterhinch, 2025-03-23:
Thanks for pointing me at this use case.

I have pushed an update documented [here](https://github.com/peterhinch/micropython-micro-gui/blob/main/README.md#42-constructor). This makes `Screen` and `Window` classes behave similarly. Please let me know if you have any problems with this.

chiefenne, 2025-03-23:
Thanks for the speedy update. It works fine. I copied the code from the screen_change.py demo.
It also helped me out in leaving the screen using async in the right way ;)

All good. Thanks again!