peterhinch/micropython-micro-gui #58: Supporting analog 2-axis joystick

bauripalash, 2025-04-16:
I am trying to use a analog 2 axis joystick with the library. Using navigation with the library mostly mention using Pins. But analog 2 axis joystick with a button uses a different thing, which is based on handling x,y coordinates and a button.

I was wondering is it possible to bind Next, Previous things to other stuff other than Pins, for example, let's assume I create a library, where joystick inputs are converted to UP, DOWN, LEFT, RIGHT and PUSH, IDLE states.

Would it be possible to trigger Next, when joystick is at, for example RIGHT state?

I hope I was able to somehow convey my idea. Please let me know your thoughts.

peterhinch, 2025-04-16:
You can buy joystick devices which consist of four switches, or five with a push action. These work well.

While it would be possible to interface a linear device, the simple switch unit is so effective that I can see no reason to adapt the GUI to this end.

bauripalash, 2025-04-16:
If I wanted to do it myself, if possible can you please point to me where should I start.