peterhinch/micropython-micro-gui #59: Where is hardware_setup.py

jhugery-blade, 2025-05-08:
So I am looking at this project and at all the examples.  There are many examples with references to hardware_setup.py.  There is no such file in this project.  Do I need to create it?  If so, how do I create it.  There is no documentation on this.

AaronMittelmeier, 2025-05-08:
Any file in setup examples can be used as your hardware_setup depending on your hardware.

peterhinch, 2025-05-08:
See [README para 1.5](https://github.com/peterhinch/micropython-micro-gui/blob/main/README.md#15-hardware-definition): you need to create `hardware_setup.py` to match your hardware. As @AaronMittelmeier says, there are many examples in the `setup_examples` directory: any of these these may be used as a template.

closed 2025-05-08 by peterhinch

jhugery-blade, 2025-05-12:
Gotcha.  Sorry I missed that.